stupidloud/nanopi-openwrt #1197: R5S RAM 4G TF无法启动？

tlb53257870, 2022-07-11:
不知道为什么 R5S 固件 写进去无法启动，我是使用TF卡启动 并不是EMMC
不知道操作是不是正确的，唯一想到的是我的RAM是4G版本

zyhusesit, 2022-07-19:
我的4G版本和你一样也是无法启动

coolqinf, 2022-07-21:
无法启动+1

trednx, 2022-07-29:
the same for me - 4GB R5S can't boot from card or if burn img to eMMC - the same problem.
can someone confirm that I shoulв use different version for 4G ?

nietsme, 2022-10-13:
一样不行，卡刷无法启动，20221012plus版，没人管吗？

klever1988, 2022-10-14:
能启动，我在用呢

himan85, 2022-10-16:
同样不能启动，SYS灯长亮

klever1988, 2022-10-16:
我是2g ram，应该是一样的呀

nietsme, 2022-10-17:
再次测试多次总结如下：
1 eMMC刷官方固件，TF卡刷官方固件。
不插卡时启动eMMC系统
插卡后启动卡里的系统。（我在固件某个地方写了备注区分是什么介质上的系统）（这种是我想使用的最完美方式）
2 eMMC刷官方固件，TF卡刷 klever或其他部分第三方固件。
不插卡时启动eMMC系统；
插卡后sys灯长亮。 经多次百度发现如下操作：“如果想要以TF卡启动此固件，需要 先用卡针捅住 MASK 按钮， 然后接入电源，3秒后松开MASK键（切勿时间长）这样将会以TF卡启动。”。     的确可以。  但MASK按钮内陷不太方便操作。   感觉如果MASK按钮正常与外壳齐平直接可以用手按，这种方式还可以接受。

疑问：为什么第三方固件没有像官方固件那样“”“不插卡走eMMC插卡走卡"""

nietsme, 2022-10-17:
> 我是2g ram，应该是一样的呀

大佬你是emmc固件清掉了    直接用的卡？  还是怎么弄的

tlb53257870, 2022-10-17:
> 重新测试组装 刷写系统：再次插入1个MMC官方，TF卡刷刷官方后启动 不启动eMMC 卡卡后提示。的系统）（这个是我想使用的完美插卡） 2 eMMC官方刷卡，TF卡刷卡 时或其他部分插入卡时不需要启动MMC系统； 多插后系统灯长亮。百度如果想要先卡这个，需要用第二次卡戳戳住MASK，然后点击开机以随后启动MASK，3次TF启动，这样才能成功启动MASK ”。确实是MASK外壳内可以方便。感觉MASK可以直接用手操作。
> 
> 疑问：为什么不需要像官方插卡那样“走eMMC”“”“”

看官方资料 启动哪里有启动方式 ，好像有很多种，有的刷到内部，就始终内部启动 ，这里可以去看官方文件，这个其实好解决 就看用那个启动方式，现在更新到什么情况了 我不清楚，当初我是用是用卡 无法启动。我怀疑就是4G版本和2G版本不一样，所以我才有这个猜测。我在用有个阿里盘的固件 ，那个没这些问题，

tlb53257870, 2022-10-17:
> 我是2g ram，应该是一样的呀

但是确实r4s 4g启动不了，我r2s用你的固件，很好用，理论上应该不是操作失误导致的启动不了，而且不只是我一个人会这样，不知道你现在 解决这个问题了没，R4S 有个阿里云盘上面的固件，不知道你知不知道，他那个固件是可以正常使用的。

xeasonwang, 2023-01-05:
> 再次测试多次总结如下： 1 eMMC刷官方固件，TF卡刷官方固件。 不插卡时启动eMMC系统 插卡后启动卡里的系统。（我在固件某个地方写了备注区分是什么介质上的系统）（这种是我想使用的最完美方式） 2 eMMC刷官方固件，TF卡刷 klever或其他部分第三方固件。 不插卡时启动eMMC系统； 插卡后sys灯长亮。 经多次百度发现如下操作：“如果想要以TF卡启动此固件，需要 先用卡针捅住 [...]

感谢，成功了

socekin, 2023-02-19:
> 再次测试多次总结如下： 1 eMMC刷官方固件，TF卡刷官方固件。 不插卡时启动eMMC系统 插卡后启动卡里的系统。（我在固件某个地方写了备注区分是什么介质上的系统）（这种是我想使用的最完美方式） 2 eMMC刷官方固件，TF卡刷 klever或其他部分第三方固件。 不插卡时启动eMMC系统； 插卡后sys灯长亮。 经多次百度发现如下操作：“如果想要以TF卡启动此固件，需要 先用卡针捅住 [...]

解决了，但是每次重启都得操作，有没有办法重启自动从TF卡启动呢