stupidloud/nanopi-openwrt #1198: 固件正常写入 正常运行两天后会出现无法写入的情况

finch-ift, 2022-07-12:
系统被设置为只读模式，无法写入文件
fsck.ext4 -y /dev/mmcblk1p2
e2fsck 1.46.5 (30-Dec-2021)
rootfs contains a file system with errors, check forced.
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Inode 117309 ref count is 1, should be 2.  Fix? yes

Unattached inode 117372
Connect to /lost+found? yes

Inode 117372 ref count is 2, should be 1.  Fix? yes

Pass 5: Checking group summary information
Free blocks count wrong for group #65 (30098, counted=30097).
Fix? yes

Free blocks count wrong (14398735, counted=14428831).
Fix? yes

rootfs: ***** FILE SYSTEM WAS MODIFIED *****
rootfs: ***** REBOOT SYSTEM *****
rootfs: 213496/3655680 files (0.0% non-contiguous), 1146721/15575552 blocks
需要手动修复才可以恢复

GreenTeodoro839, 2022-07-12:
换sd卡/尽量避免断电，ext4的通病，容易变只读

finch-ift, 2022-07-13:
> 换sd卡/尽量避免断电，ext4的通病，容易变只读

用的是闪迪的至尊超极速TF卡 还能换什么卡

finch-ift, 2022-07-13:
> 用 squashfs 格式就不会出现这样的问题

请问怎么刷squashfs 格式

zhangguanzhang, 2022-07-13:
> > 用 squashfs 格式就不会出现这样的问题
> 
> 请问怎么刷squashfs 格式

https://github.com/zhangguanzhang/Actions-OpenWrt  我的项目，支持squashfs完全识别分区扩容和在线升级，本地源ipk比1988的还多

finch-ift, 2022-07-13:
> > > 用 squashfs 格式就不会出现这样的问题
> > 
> > 
> > 请问怎么刷squashfs 格式
> 
> https://github.com/zhangguanzhang/Actions-OpenWrt 我的项目，支持squashfs完全识别分区扩容和在线升级，本地源ipk比1988的还多，还用更多源码编译

这个有点麻烦，我已经配置好了，重新刷机有点麻烦，我也找到了解决方案，系统启动后检查文件系统是否只读，如果只读就修复然后重启

Kimaru-bili, 2022-07-15:
> > > 用 squashfs 格式就不会出现这样的问题
> > 
> > 
> > 请问怎么刷squashfs 格式
> 
> https://github.com/zhangguanzhang/Actions-OpenWrt 我的项目，支持squashfs完全识别分区扩容和在线升级，本地源ipk比1988的还多，还用更多源码编译

这个刷完LAN口用不了 进不去后台 不知道怎么回事 R4S1G版

Kimaru-bili, 2022-07-15:
> 

获取不到 我打算再刷这个了 这个用在线升级就会开不开机 不升级还没法扩容 唉

Kimaru-bili, 2022-07-15:
> > > > > 用 squashfs 格式就不会出现这样的问题
> > > > 
> > > > 
> > > > 请问怎么刷squashfs 格式
> > > 
> > > 
> > > https://github.com/zhangguanzhang/Actions-OpenWrt 我的项目，支持squashfs完全识别分区扩容和在线升级，本地源ipk比1988的还多，还用更多源码编译
> > 
> > 
> > 这个刷完LAN口用不了 进不去后台 不知道怎么回事 R4S1G版
> 
> 能获取到ip吗，ipconfig看下，再不行试试天灵版本编译的

天灵版本？emmmm我不太懂 你那个里边只有test有包 我就用那个了 刷完sys灯常亮 LAN口不好使 lede silm版

zhangguanzhang, 2022-07-15:
> > > > > > 用 squashfs 格式就不会出现这样的问题
> > > > > 
> > > > > 
> > > > > 请问怎么刷squashfs 格式
> > > > 
> > > > 
> > > > https://github.com/zhangguanzhang/Actions-OpenWrt 我的项目，支持squashfs完全识别分区扩容和在线升级，本地源ipk比1988的还多，还用更多源码编译
> > > 
> > > 
> > > 这个刷完LAN口用不了 进不去后台 不知道怎么回事 R4S1G版
> > 
> > 
> > 能获取到ip吗，ipconfig看下，再不行试试天灵版本编译的
> 
> 天灵版本？emmmm我不太懂 你那个里边只有test有包 我就用那个了 刷完sys灯常亮 LAN口不好使 lede slim版

imm那个关键字的就是天灵的，你是r2s还是r4s，是squashfs的吗

Kimaru-bili, 2022-07-15:
再action里看到imm了 等会我试试

Kimaru-bili, 2022-07-15:
> 再action里看到imm了 等会我试试

只有18.06么（小声 想用最新

Kimaru-bili, 2022-07-15:
额
[image]
好像 依旧 没有获取到IP

Kimaru-bili, 2022-07-15:
这回wan口也寄了 接在路由器上灯根本不亮 LAN口接在电脑上也不亮

zhangguanzhang, 2022-07-15:
你刷的文件是哪个，或者从我博客加下我微信，我帮你看下

Kimaru-bili, 2022-07-15:
[image]

Kimaru-bili, 2022-07-15:
> [image]

这个

Kimaru-bili, 2022-07-15:
> 你刷的文件是哪个，或者从我博客加下我微信，我帮你看下
> […](#)

OK

GreenTeodoro839, 2022-07-15:
> > 再action里看到imm了 等会我试试
> 
> 只有18.06么（小声 想用最新

高版本挂载外部overlay不一定成功，18.6那个修复了

GreenTeodoro839, 2022-07-15:
> > > > > > 用 squashfs 格式就不会出现这样的问题
> > > > > 
> > > > > 
> > > > > 请问怎么刷squashfs 格式
> > > > 
> > > > 
> > > > https://github.com/zhangguanzhang/Actions-OpenWrt 我的项目，支持squashfs完全识别分区扩容和在线升级，本地源ipk比1988的还多，还用更多源码编译
> > > 
> > > 
> > > 这个刷完LAN口用不了 进不去后台 不知道怎么回事 R4S1G版
> > 
> > 
> > 能获取到ip吗，ipconfig看下，再不行试试天灵版本编译的
> 
> 天灵版本？emmmm我不太懂 你那个里边只有test有包 我就用那个了 刷完sys灯常亮 LAN口不好使 lede slim版

imm分支我觉得小苏的最好，在线软件源+本地kmod，不会出现内核不兼容，也有squashfs格式的
https://github.com/SuLingGG/OpenWrt-Rpi

zhangguanzhang, 2022-07-15:
> imm分支我觉得小苏的最好，在线软件源+本地kmod，不会出现内核不兼容，也有squashfs格式的 https://github.com/SuLingGG/OpenWrt-Rpi

但是我的升级脚本能squashfs在线扩容，等我把这个开机的问题解决吧，下面这个是我squashfs的信息：
```
root@OpenWrt:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0 58.7G  0 loop /overlay
mmcblk0     179:0    0 59.1G  0 disk 
├─mmcblk0p1 179:1    0   16M  0 part /mnt/mmcblk0p1
└─mmcblk0p2 179:2    0 59.1G  0 part /rom
root@OpenWrt:~# df -h
Filesystem          Size  Used Avail Use% Mounted on
/dev/root           375M  375M     0 100% /rom
tmpfs               494M  136K  494M   1% /tmp
/dev/loop0           59G   16G   44G  26% /overlay
overlayfs:/overlay   59G   16G   44G  26% /
tmpfs               512K     0  512K   0% /dev
/dev/mmcblk0p1       16M   15M  524K  97% /mnt/mmcblk0p1
```

zhangguanzhang, 2022-07-15:
问题已经解决了，r4s我少了网卡驱动，已经给他整好了

closed 2022-07-22 by finch-ift

zhangguanzhang, 2022-07-26:
@Kimaru-bili 上下微信

choicky, 2022-09-26:
> > > > 用 squashfs 格式就不会出现这样的问题
> > > 
> > > 
> > > 请问怎么刷squashfs 格式
> > 
> > 
> > https://github.com/zhangguanzhang/Actions-OpenWrt 我的项目，支持squashfs完全识别分区扩容和在线升级，本地源ipk比1988的还多，还用更多源码编译
> 
> 这个有点麻烦，我已经配置好了，重新刷机有点麻烦，我也找到了解决方案，系统启动后检查文件系统是否只读，如果只读就修复然后重启

@finch-ift 这一步是如何操作呢？能自动化进行吗？

zhangguanzhang, 2022-10-11:
test的release是编译上传好的（可以走ghproxy下载），action里的可能下载不稳定