stupidloud/nanopi-openwrt #1256: 在线升级报错

whuanghong, 2022-11-14:
root@OpenWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | bash
Downloading https://mirrors.tencent.com/lede/snapshots/targets/x86/64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package coreutils-truncate (9.1-1) installed in root is up to date.
准备更新到2022-11-11
--2022-11-14 09:00:19--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2022-11-11/x86.img.gz
Resolving ghproxy.com... 129.146.41.82
Connecting to ghproxy.com|129.146.41.82|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2022-11-14 09:00:21 ERROR 404: NOT FOUND.

gzip: invalid magic
awk: md5sum.txt: No such file or directory

lonma, 2022-11-15:
这个opkg的连接可以加上类似src/gz immortalwrt_telephony https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/management这种么？我直接ttyd下opkg install docker提示无法安装，是操作方法不对么

H-Tung, 2022-12-29:
https://github.com/stupidloud/nanopi-openwrt/issues/1191#issuecomment-1172999050