stupidloud/nanopi-openwrt #1260: 固件软件源错了

wittcong, 2022-11-25:
软件源还是天灵的，版本不匹配

klever1988, 2022-11-25:
软件源名字是immortal而已，软件源地址都是op官方snapshot的腾讯镜像

wittcong, 2022-11-26:
> 软件源名字是immortal而已，软件源地址都是op官方snapshot的腾讯镜像

我安装了几个包，从lean的软件源拉取的可以正常安装，内置的虽然安装成功，但是不工作，luci都出不来……

wittcong, 2022-11-26:
src/gz local file:///local_feed
#src/gz openwrt_base https://mirrors.cloud.tencent.com/openwrt/releases/18.06.8/packages/aarch64_generic/base
#src/gz openwrt_luci https://mirrors.cloud.tencent.com/openwrt/releases/18.06.8/packages/aarch64_generic/luci
#src/gz openwrt_packages https://mirrors.cloud.tencent.com/openwrt/releases/18.06.8/packages/aarch64_generic/packages
#src/gz openwrt_routing https://mirrors.cloud.tencent.com/openwrt/releases/18.06.8/packages/aarch64_generic/routing
#src/gz openwrt_telephony https://mirrors.cloud.tencent.com/openwrt/releases/18.06.8/packages/aarch64_generic/telephony