stupidloud/nanopi-openwrt #1262: fork后，自定义添加插件 `luci-app-frpc`失败

coppea, 2022-12-02:
在R4S的seed添加`luci-app-frpc=y`编译后在左侧服务没有显示，后手动安装成功还是没有显示
以下是自定义插件配置
CONFIG_PACKAGE_default-settings-chn=y
CONFIG_PACKAGE_kmod-ebtables-ipv6=m
CONFIG_PACKAGE_kmod-ipt-raw6=m
CONFIG_PACKAGE_kmod-usb-net-asix-ax88179=m
CONFIG_PACKAGE_kmod-usb-net-rndis=m
CONFIG_PACKAGE_ksmbd-utils=m
CONFIG_PACKAGE_luci-app-mwan3=m
CONFIG_PACKAGE_luci-app-mwan3helper=m
CONFIG_PACKAGE_luci-app-syncdial=m
CONFIG_PACKAGE_luci-app-wireguard=m
CONFIG_PACKAGE_luci-proto-bonding=m
CONFIG_PACKAGE_luci-proto-openconnect=m
CONFIG_PACKAGE_cifsmount=m
CONFIG_PACKAGE_ebtables=m
CONFIG_PACKAGE_fstrim=m
CONFIG_PACKAGE_igmpproxy=m
CONFIG_PACKAGE_node-npm=m
CONFIG_PACKAGE_shadowsocks-libev-ss-tunnel=m
CONFIG_PACKAGE_smartmontools-drivedb=m
CONFIG_PACKAGE_usbutils=m
CONFIG_PACKAGE_xl2tpd=m
CONFIG_PACKAGE_zram-swap=m

#这些是会预装到“胖版”固件，并且也会预置到slim版软件源的配置
CONFIG_CCACHE=y
CONFIG_DEVEL=y
CONFIG_IB_STANDALONE=y
CONFIG_IB=y
CONFIG_TARGET_OPTIONS=y
CONFIG_TARGET_ROOTFS_EXT4FS=y
CONFIG_TARGET_IMAGES_GZIP=y
#CONFIG_TARGET_ROOTFS_SQUASHFS is not set
#CONFIG_OPENSSL_ENGINE_BUILTIN_AFALG is not set
#CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO is not set
#CONFIG_SIGNED_PACKAGES is not set
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUSYBOX_CONFIG_UNICODE_SUPPORT=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CHECK_UNICODE_IN_ENV=y
CONFIG_DROPBEAR_ECC_FULL=y
CONFIG_DROPBEAR_ECC=y
CONFIG_PACKAGE_luci-app-aria2=y
CONFIG_PACKAGE_luci-app-arpbind=y
CONFIG_PACKAGE_luci-app-autoreboot=y
CONFIG_PACKAGE_luci-app-cifs-mount=y
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-filetransfer=y
CONFIG_PACKAGE_luci-app-ksmbd=y
CONFIG_PACKAGE_luci-app-services-wolplus=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_shortcut-fe=y
CONFIG_PACKAGE_luci-app-turboacc=y
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-app-usb-printer=y
CONFIG_PACKAGE_luci-app-v2raya=y
CONFIG_PACKAGE_luci-app-vsftpd=y
CONFIG_PACKAGE_luci-lib-json=y
CONFIG_PACKAGE_luci-mod-rpc=y
CONFIG_PACKAGE_luci-theme-argonv3=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_luci-theme-infinityfreedom=y
CONFIG_PACKAGE_luci-theme-material=y
CONFIG_PACKAGE_luci-theme-netgear=y
CONFIG_PACKAGE_luci-theme-rosy=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_coremark=y
CONFIG_PACKAGE_coreutils-dd=y
CONFIG_PACKAGE_coreutils-nohup=y
CONFIG_PACKAGE_coreutils-truncate=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_iptables-mod-conntrack-extra=y
CONFIG_PACKAGE_iptables-mod-ipopt=y
CONFIG_PACKAGE_iptables-mod-tproxy=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_kmod-fast-classifier=y
CONFIG_PACKAGE_kmod-fs-f2fs=y
CONFIG_PACKAGE_kmod-ipt-offload=y
CONFIG_PACKAGE_kmod-pppol2tp=y
CONFIG_PACKAGE_kmod-tcp-bbr=y
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_kmod-usb3=y
CONFIG_PACKAGE_kmod-usb-net-rtl8152-vendor=y
CONFIG_PACKAGE_losetup=y
CONFIG_PACKAGE_openssh-sftp-server=y
CONFIG_PACKAGE_pv=y
CONFIG_PACKAGE_resize2fs=y
CONFIG_PACKAGE_sfdisk=y
CONFIG_PACKAGE_snmpd=y
CONFIG_PACKAGE_upx=y
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_wget=y

CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r4s=y

CONFIG_TARGET_OPTIMIZATION="-O3 -pipe -march=armv8-a+crypto+crc -mcpu=cortex-a72.cortex-a53+crypto+crc -mtune=cortex-a72.cortex-a53"
CONFIG_TARGET_ROOTFS_PARTSIZE=600

CONFIG_PACKAGE_luci-app-dockerman=y
CONFIG_DOCKER_OPTIONAL_FEATURES=y
CONFIG_PACKAGE_luci-app-openclash=y
CONFIG_PACKAGE_luci-app-cpufreq=y
CONFIG_PACKAGE_luci-app-fileassistant=y
CONFIG_PACKAGE_luci-app-qbittorrent=y
CONFIG_PACKAGE_luci-app-statistics=y
CONFIG_PACKAGE_luci-app-uugamebooster=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-theme-openwrt=y
CONFIG_PACKAGE_luci-theme-openwrt-2020=y

CONFIG_PACKAGE_kmod-gpu-lima=y
CONFIG_PACKAGE_kmod-ath9k-htc=y
CONFIG_PACKAGE_kmod-mt76x0u=y
CONFIG_PACKAGE_kmod-mt76x2u=y
CONFIG_PACKAGE_kmod-r8125=y
CONFIG_PACKAGE_kmod-rtw88=y

CONFIG_PACKAGE_iw=y
CONFIG_PACKAGE_iwinfo=y
CONFIG_PACKAGE_wpad-wolfssl=y
CONFIG_DRIVER_11AC_SUPPORT=y
CONFIG_DRIVER_11N_SUPPORT=y
CONFIG_DRIVER_11W_SUPPORT=y

coppea, 2022-12-02:
不支持此版本frpc，http://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/luci-app-frpc_git-22.229.49012-175487f_all.ipk，安装成功，但是在服务不显示

不过支持基本不维护的旧版本frpc ，https://github.com/kuoruan/luci-app-frpc，在服务能显示

能否查看一下是什么原因不支持frpc？

john5du, 2022-12-06:
试试 CONFIG_PACKAGE_luci-app-frpc=y

coppea, 2022-12-09:
能否添加luci-app-frpc支持

closed 2022-12-16 by coppea