stupidloud/nanopi-openwrt #1307: 更新脚本每次都比较久

AiharaMahiru, 2023-08-07:
BusyBox v1.36.0 (2023-07-31 14:06:00 UTC) built-in shell (ash)
     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|
 \        \   DE /
  \    LE  \    /  -------------------------------------------
   \  DE    \  /    OpenWrt SNAPSHOT, r6123-2abe7192f
    \________\/    -------------------------------------------
root@OpenWrt:~# wget -qO- https://github.com/stupidloud/nanopi-openwrt/raw/master/s
cripts/autoupdate-bash.sh | ver=-slim bash
Downloading file:///local_feed/Packages.gz
Updated list of available packages in /var/opkg-lists/local
Package coreutils-truncate (9.1-1) installed in root is up to date.
准 备 更 新 到 2023-08-07
--2023-08-08 00:05:47--  https://ghproxy.com/https://github.com/stupidloud/nanopi-openwrt/releases/download/2023-08-07/r2s-slim.img.gz
Resolving ghproxy.com... 192.9.132.155
Connecting to ghproxy.com|192.9.132.155|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 374200392 (357M) [application/octet-stream]
Saving to: 'STDOUT'
-                   100%[===================>] 356.87M   599KB/s    in 10m 11s
2023-08-08 00:16:00 (598 KB/s) - written to stdout [374200392/374200392]
--2023-08-08 00:16:00--  https://ghproxy.com/https://github.com/stupidloud/nanopi-openwrt/releases/download/2023-08-07/r2s-slim.img.gz.md5
Resolving ghproxy.com... 192.9.132.155
Connecting to ghproxy.com|192.9.132.155|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 50 [application/octet-stream]
Saving to: 'md5sum.txt'
md5sum.txt          100%[===================>]      50  --.-KB/s    in 0s
2023-08-08 00:16:02 (5.64 MB/s) - 'md5sum.txt' saved [50/50]
2023-08-07固 件 已 下 载
Checking that no-one is using this disk right now ... OK
Disk FriendlyWrt.img: 233.2 GiB, 250399948800 bytes, 489062400 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5452574f
Old situation:
Device           Boot  Start     End Sectors  Size Id Type
FriendlyWrt.img1 *     65536  131071   65536   32M 83 Linux
FriendlyWrt.img2      196608 1507327 1310720  640M 83 Linux
FriendlyWrt.img2:
New situation:
Disklabel type: dos
Disk identifier: 0x5452574f
Device           Boot  Start       End   Sectors   Size Id Type
FriendlyWrt.img1 *     65536    131071     65536    32M 83 Linux
FriendlyWrt.img2      196608 489062399 488865792 233.1G 83 Linux
The partition table has been altered.
Syncing disks.
解 压 已 完 成 ， 准 备 编 辑 镜 像 文 件 ， 写 入 备 份 信 息
Tue Aug  8 00:16:12 CST 2023 upgrade: Saving config files...
备 份 文 件 已 经 写 入 ， 移 除 挂 载
e2fsck 1.46.5 (30-Dec-2021)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Padding at end of inode bitmap is not set. Fix? yes
rootfs: ***** FILE SYSTEM WAS MODIFIED *****
rootfs: 3618/40960 files (0.0% non-contiguous), 106554/163840 blocks
resize2fs 1.46.5 (30-Dec-2021)
Resizing the filesystem on /dev/loop0p2 to 61108224 (4k) blocks.
The filesystem on /dev/loop0p2 is now 61108224 (4k) blocks long.
正 在 打 包 ...
开 始 写 入 ， 请 勿 中 断 ...
60554215424 bytes (61 GB, 56 GiB) copied, 217 s, 279 MB/s

edisonchan, 2023-08-17:
你们的 TF 卡刷了后有做 ext4 分区扩容吗？看这个脚本，更新后似乎有做重写分区的动作，那是不是意味着在线更新会导致 ext4 分区扩容设置失效？

stupidloud, 2023-08-17:
> 你们的 TF 卡刷了后有做 ext4 分区扩容吗？看这个脚本，更新后似乎有做重写分区的动作，那是不是意味着在线更新会导致 ext4 分区扩容设置失效？

正常更新完成后会自动重新扩容好并且重启

AiharaMahiru, 2023-08-17:
> > 你们的 TF 卡刷了后有做 ext4 分区扩容吗？看这个脚本，更新后似乎有做重写分区的动作，那是不是意味着在线更新会导致 ext4 分区扩容设置失效？
> 
> 正常更新完成后会自动重新扩容好并且重启

问题就在这，我刷完slim之后，手动扩容挂载了，然后使用在线升级每次都过一遍内存卡，害怕内存卡被干报废

edisonchan, 2023-08-19:
> > 你们的 TF 卡刷了后有做 ext4 分区扩容吗？看这个脚本，更新后似乎有做重写分区的动作，那是不是意味着在线更新会导致 ext4 分区扩容设置失效？
> 
> 正常更新完成后会自动重新扩容好并且重启

我这里 r2s 使用 docker 固件，在 ssh 里执行自动更新脚本，提示执行下载股剑道 70% 左右就会丢失 ssh 连接，然后也无法上网，虽然 DHCP 服务似乎还在运作（能自动分配 IP），等了 10 分钟后还是不能连接 ssh 和上网，只能拔电源线，然后重启后看到的依然是之前的固件在运作。