stupidloud/nanopi-openwrt #1308: 请问有R6s的支持计划吗

zhrui0517, 2023-08-17:


stupidloud, 2023-08-22:
发了 试试看

zhangsunyucong, 2023-08-25:
R6C可以用吗

zhrui0517, 2023-09-01:
> 发了 试试看

试了一下发现几个bug：
1、eth1和eth2绑定关系和硬件是反的
2、首页概览页没有各网口的使用状态
3、TURBOACC里没有Flow Offloading只有Shortcut FE，当然这个有没有实际区别我不是很清楚。

目前没发现别的

stupidloud, 2023-09-01:
1,2, 已修复
3 那个设置没什么实际作用

stupidloud, 2023-09-02:
> R6C可以用吗

可以 试试

zhangsunyucong, 2023-09-02:
发现nvme协议，m2协议固态硬盘好像无法识别

zhrui0517, 2023-09-05:
> 1,2, 已修复 3 那个设置没什么实际作用

> 1,2, 已修复 3 那个设置没什么实际作用

1、你好，我看到了code里这条sed -i "s/ucidef_set_interfaces_lan_wan 'eth0 eth1' 'eth2'/ucidef_set_interfaces_lan_wan 'eth1 eth2' 'eth0'/" target/linux/rockchip/armv8/base-files/etc/board.d/02_network

但实际上不是lan/wan口绑定错误，目前的问题是固件认为的eth1口实际上是主板上的eth2口，eth2口对应主板上的eth1口；
R6s的wan口应该就是eth2口。

2、第二个问题是我之前用作者r4、r2的固件，首页都有一部分显示各个网口的状态，是否接入、协商速率等信息，但目前到9月2号R6s的固件还是没有这部分信息的。

stupidloud, 2023-09-09:
> 1、你好，我看到了code里这条sed -i "s/ucidef_set_interfaces_lan_wan 'eth0 eth1' 'eth2'/ucidef_set_interfaces_lan_wan 'eth1 eth2' 'eth0'/" target/linux/rockchip/armv8/base-files/etc/board.d/02_network [...]

测试一下这两个问题是否修复了

zhrui0517, 2023-11-11:
> 测试一下这两个问题是否修复了

第二个问题修复了。
第一个没有，我的解决方案是，wan绑定eth1，lan绑定eth0/2；然后LED里把eth1和eth2对调就好了。这样除了首页三个网口显示能看出来eth1/2是反的，其他都正常了。我看kiddin9对于R6s的解决方案大概也是这样，貌似除了lean自己发布的R6s固件对应是正常的。