stupidloud/nanopi-openwrt #1310: r2s最新完整版固件无法识别811无线网卡

piexller, 2023-08-25:
r2s最新完整版固件无法识别cf-811ac无线网卡，网卡芯片为rt8811.
[image]

weiqiye, 2023-12-18:
> 用官方最新镜像（OpenWrt 22.03.5 r20134-5f15225c1e）可以驱动cf-811ac的单频AP模式，但不能同时

你好，能分享下这个版本的固件吗？官网找不到这个固件了，另外这个固件无线带宽能到433吗？

ztatvialto, 2023-12-18:
> > 用官方最新镜像（OpenWrt 22.03.5 r20134-5f15225c1e）可以驱动cf-811ac的单频AP模式，但不能同时
> 
> 你好，能分享下这个版本的固件吗？官网找不到这个固件了，另外这个固件无线带宽能到433吗？

不知道用的哪个版本了，不是什么特殊的，可能就是8月底的最新版

ztatw, 2023-12-18:
> > 用官方最新镜像（OpenWrt 22.03.5 r20134-5f15225c1e）可以驱动cf-811ac的单频AP模式，但不能同时
> 
> 你好，能分享下这个版本的固件吗？官网找不到这个固件了，另外这个固件无线带宽能到433吗？

不知道用的哪个版本了，不是什么特殊的，可能就是8月底的最新版

weiqiye, 2023-12-18:
> > > 用官方最新镜像（OpenWrt 22.03.5 r20134-5f15225c1e）可以驱动cf-811ac的单频AP模式，但不能同时
> > 
> > 你好，能分享下这个版本的固件吗？官网找不到这个固件了，另外这个固件无线带宽能到433吗？
> 
> 不知道用的哪个版本了，不是什么特殊的，可能就是8月底的最新版

带宽能到433M吗？今年8月底吗？目前官方最新版带宽只能到150M

ztatw, 2023-12-18:
> > > > 用官方最新镜像（OpenWrt 22.03.5 r20134-5f15225c1e）可以驱动cf-811ac的单频AP模式，但不能同时
> > > 
> > > 
> > > 你好，能分享下这个版本的固件吗？官网找不到这个固件了，另外这个固件无线带宽能到433吗？
> > 
> > 
> > 不知道用的哪个版本了，不是什么特殊的，可能就是8月底的最新版
> 
> 带宽能到433M吗？今年8月底吗？目前官方最新版带宽只能到150M

没太注意性能，能驱动起来已经不容易了。后来直接买个小米路由器当有线AP了，目前手里有2个小米千兆版待售。。。

will-huang9531, 2024-07-27:
> > > > 用官方最新镜像（OpenWrt 22.03.5 r20134-5f15225c1e）可以驱动cf-811ac的单频AP模式，但不能同时
> > > 
> > > 
> > > 你好，能分享下这个版本的固件吗？官网找不到这个固件了，另外这个固件无线带宽能到433吗？
> > 
> > 
> > 不知道用的哪个版本了，不是什么特殊的，可能就是8月底的最新版
> 
> 带宽能到433M吗？今年8月底吗？目前官方最新版带宽只能到150M

 请问有新进展吗？CF-811AC目前能买到的是V3版本的，网卡是RTL8811CU，只有官方的两个固件能驱动起来，但是这两个固件开master模式都只能握手802.11n模式的40mhz，顶天150M，没有用什么办法能跑80mhz的？

freehuoshan, 2024-10-29:
> 请问有新进展吗？CF-811AC目前能买到的是V3版本的，网卡是RTL8811CU，只有官方的两个固件能驱动起来，但是这两个固件开master模式都只能握手802.11n模式的40mhz，顶天150M，没有用什么办法能跑80mhz的？

官方的哪两个固件可以驱动 CF-811AC V3 ？ 谢谢.