stupidloud/nanopi-openwrt #1313: 自恢复更新以来R1S H5 Slim版本都不行

gekkjoe, 2023-09-03:
自恢复更新之后的R1S H5 Slim版本我都试过了 都是开不了机 期待作者跟进

stupidloud, 2023-09-03:
这个呢

https://github.com/stupidloud/nanopi-openwrt/releases/download/2023-09-02/r1s.img.gz

gekkjoe, 2023-09-03:
> 这个呢
> 
> https://github.com/stupidloud/nanopi-openwrt/releases/download/2023-09-02/r1s.img.gz

非Slim的我只试过最新的 就是902这个 也是不行 (感谢回复)

stupidloud, 2023-09-05:
可以从红色指示灯判断启动是否成功吗？

gekkjoe, 2023-09-06:
> 可以从红色指示灯判断启动是否成功吗？

最新0905仍然不行哦  红灯虽然没有闪烁但是网线连接电脑显示"未识别网络" 关键是自带WiFi没有出来 正常版本就算网线不行但是正常的话wifi肯定会出来的

gekkjoe, 2023-09-09:
> 可以从红色指示灯判断启动是否成功吗？

0908的R1S固件仍然不行

PillarsZhang, 2023-09-11:
也是R1S H5，我在用的2023-08-28非slim版正常，但slim版也是上电sys常亮后再闪烁（如果没记错
于是...我把slim版的local_feeds文件夹给移植过去了，再改改源🤣
非slim版如果无法启动可以断电再上电/插拔SD卡/拿风扇怼着吹试试
另外现在版本固件似乎默认不开wifi了？我看着没有

stupidloud, 2023-09-11:
感觉好像是软件源文件太多，包太大导致的，我精简掉一些再试试

gekkjoe, 2023-09-12:
> 也是R1S H5，我在用的2023-08-28非slim版正常，但slim版也是上电sys常亮后再闪烁（如果没记错 于是...我把slim版的local_feeds文件夹给移植过去了，再改改源🤣 非slim版如果无法启动可以断电再上电/插拔SD卡/拿风扇怼着吹试试 另外现在版本固件似乎默认不开wifi了？我看着没有

谢谢帮忙 0828的两个版本都试过了 插网线显示未识别网络 重启几次还是不行 我用2022的10月以前的某个固件才会正常
[image]
这是挑出来的 有些测试过不正常忘记删了 现在也忘记哪些是正常

> 感觉好像是软件源文件太多，包太大导致的，我精简掉一些再试试

0912也试过了 也是不行 红灯长亮 LAN口插电脑显示未识别网络

peterxutong, 2023-09-15:
> 感觉好像是软件源文件太多，包太大导致的，我精简掉一些再试试

对比了下，7月25日的编译文件采用test_kernel ，r1s的包就不正常了，默认没有WiFi模块
大佬有空检查下是啥问题呢

gekkjoe, 2023-09-16:
> 感觉好像是软件源文件太多，包太大导致的，我精简掉一些再试试

915的R1SH5固件还是不行 红灯长亮但是网线连接电脑显示未识别网络

gekkjoe, 2023-09-17:
> 感觉好像是软件源文件太多，包太大导致的，我精简掉一些再试试

917的固件能正常进入系统各种设置了 就是软件源没有 于是自己找了一个 但是更新内核的时候不行 奇怪了 下载也不行 因为路径是个空路径

gekkjoe, 2023-09-29:
> 感觉好像是软件源文件太多，包太大导致的，我精简掉一些再试试

928使用正常!

gekkjoe, 2023-09-30:
> 也是R1S H5，我在用的2023-08-28非slim版正常，但slim版也是上电sys常亮后再闪烁（如果没记错 于是...我把slim版的local_feeds文件夹给移植过去了，再改改源🤣 非slim版如果无法启动可以断电再上电/插拔SD卡/拿风扇怼着吹试试 另外现在版本固件似乎默认不开wifi了？我看着没有

928使用正常!

gekkjoe, 2023-09-30:
> > 感觉好像是软件源文件太多，包太大导致的，我精简掉一些再试试
> 
> 对比了下，7月25日的编译文件采用test_kernel ，r1s的包就不正常了，默认没有WiFi模块 大佬有空检查下是啥问题呢

928使用正常!

BorderlineWei, 2023-10-17:
> > > 感觉好像是软件源文件太多，包太大导致的，我精简掉一些再试试
> > 
> > 
> > 对比了下，7月25日的编译文件采用test_kernel ，r1s的包就不正常了，默认没有WiFi模块 大佬有空检查下是啥问题呢
> 
> 928使用正常!

有wifi了吗r1s