stupidloud/nanopi-openwrt #1340: 安装不了fuse3,不知道为什么，会提示如下错误

Joker-lm, 2024-01-02:
用opkg update
opkg install fuse3-utils
命令安装fuse3时会出现如下错误

* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.176-1-c90b8cdeba326c5459f96b8e3bd16c5a) for kmod-fuse
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fuse found, but incompatible with the architectures configured

stupidloud, 2024-01-03:
今天开始内置了

Joker-lm, 2024-01-03:
> 今天开始内置了

今天发布的这个版本已经内置了吗？感谢大佬

stupidloud, 2024-01-05:
有的
[image]

Joker-lm, 2024-01-05:
> 有的 [...]
已经用上了，感谢大佬