stupidloud/nanopi-openwrt #1342: 最新固件（2024-01-08）中，是否已经取消了Hello World（vssr）？

calkchu, 2024-01-10:
安装最新固件slim版本，本地源中没有搜索到HelloWorld应用，搜索vssr搜索不到，是否已经取消了呢？自己安装该如何安装呢？大神能否告知一下，感谢！

stupidloud, 2024-01-10:
因为原作者删库了，所以也没办法编译
https://github.com/jerrykuku/luci-app-vssr

calkchu, 2024-01-11:
> 因为原作者删库了，所以也没办法编译 https://github.com/jerrykuku/luci-app-vssr

了解了，请问大神，除了passwall外，能否推荐一款支持hysteria2协议的插件呢？

gowy222, 2024-01-21:
> > 因为原作者删库了，所以也没办法编译 https://github.com/jerrykuku/luci-app-vssr
> 
> 了解了，请问大神，除了passwall外，能否推荐一款支持hysteria2协议的插件呢？

sing-box 天下无敌...涵盖所有协议~

calkchu, 2024-01-22:
> > > 因为原作者删库了，所以也没办法编译 https://github.com/jerrykuku/luci-app-vssr
> > 
> > 
> > 了解了，请问大神，除了passwall外，能否推荐一款支持hysteria2协议的插件呢？
> 
> sing-box 天下无敌...涵盖所有协议~

好的，我试试看，感谢

stupidloud, 2024-01-23:
> > > 因为原作者删库了，所以也没办法编译 https://github.com/jerrykuku/luci-app-vssr
> > 
> > 
> > 了解了，请问大神，除了passwall外，能否推荐一款支持hysteria2协议的插件呢？
> 
> sing-box 天下无敌...涵盖所有协议~

https://github.com/immortalwrt/homeproxy

这个吗？
有没有什么版本要求？
firewall4什么的

calkchu, 2024-01-24:
> > > > 因为原作者删库了，所以也没办法编译 https://github.com/jerrykuku/luci-app-vssr
> > > 
> > > 
> > > 了解了，请问大神，除了passwall外，能否推荐一款支持hysteria2协议的插件呢？
> > 
> > 
> > sing-box 天下无敌...涵盖所有协议~
> 
> https://github.com/immortalwrt/homeproxy
> 
> 这个吗？ 有没有什么版本要求？ firewall4什么的

我还真不知道，不过我没啥要求，版本也没有Firewall4是不是现在都用呀？还是默认屏蔽掉？我也不知道

stupidloud, 2024-01-24:
@gowy222