stupidloud/nanopi-openwrt #1351: passwall缺少依赖

latel, 2024-03-16:
开箱后啥都没搞，直接配置就报错了。。。

2024-03-17 05:06:06: 开始订阅...
2024-03-17 05:06:06: 正在订阅:【vultr】https://api.v1.mk/sub?xxxxxxxx
2024-03-17 05:06:07: 成功解析【vultr】节点数量: 1
2024-03-17 05:06:07: 订阅完毕...
2024-03-17 05:06:11: 清空并关闭相关程序和缓存完成。
2024-03-17 05:06:11: iptables透明代理基础依赖 iptables-mod-socket 未安装...
2024-03-17 05:06:11: iptables透明代理基础依赖 iptables-mod-iprange 未安装...
2024-03-17 05:06:11: 运行完成！

latel, 2024-03-16:
去系统-软件包里手动安装也失败。。。。

```
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Installing iptables-mod-socket (1.8.8-2) to root...
Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/iptables-mod-socket_1.8.8-2_aarch64_generic.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.81-1-ecd012334be04f6a499f3f7710bd8c12) for kmod-nf-socket
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-socket found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.81-1-ecd012334be04f6a499f3f7710bd8c12) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.81-1-ecd012334be04f6a499f3f7710bd8c12) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.81-1-ecd012334be04f6a499f3f7710bd8c12) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.81-1-ecd012334be04f6a499f3f7710bd8c12) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.81-1-ecd012334be04f6a499f3f7710bd8c12) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.81-1-ecd012334be04f6a499f3f7710bd8c12) for kmod-ipt-socket
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-socket found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for iptables-mod-socket:
 * 	kernel (= 6.1.81-1-ecd012334be04f6a499f3f7710bd8c12)
 * opkg_install_cmd: Cannot install package iptables-mod-socket.
```

latel, 2024-03-16:
终端也打不开 😢 
[image]

latel, 2024-03-16:
重启路由后，终端菜单直接消失了 😢 
[image]

stupidloud, 2024-03-18:
更新了，你等编译完成再试试

latel, 2024-03-19:
> 更新了，你等编译完成再试试

可以知道下怎么升级吗，下载安装包后 webui里上传压缩包点升级，还是重新dd刷tf卡呀？

latel, 2024-04-19:
@stupidloud 大佬怎么升级下，我看UI里升级要用sysupgrade包，但是每个版本的release文件里没看到有， 
另外试了终端内升级，但是报错了https://github.com/stupidloud/nanopi-openwrt/issues/1364