stupidloud/nanopi-openwrt #1360: 小白能否请教下要怎么扩容个分区？

latel, 2024-04-07:
[image]

docker的分区太小了，我随便装了个traefik就提示磁盘存储不够。我需要把docker的分区调的大一些。

```bash
root@OpenWrt:~# docker pull traefik:latest
latest: Pulling from library/traefik
bca4290a9639: Pull complete
3d4baaba5180: Pull complete
54ec8d4c0d6b: Extracting [==================================================>]  37.69MB/37.69MB
eb9404f9e59a: Download complete
failed to register layer: write /usr/local/bin/traefik: no space left on device
root@OpenWrt:~#
```

stupidloud, 2024-04-07:
最简单的方法是卡拔下来通过读卡器连接到运行linux或者macos的电脑上，用fdisk分区工具和resize2fs进行重建分区和扩容

https://www.cnblogs.com/gucb/p/11528507.html

hza1128, 2024-04-08:
在你的磁盘那点修改，然后拉到下面:如图，然后挂载为 docker数据分区，对应修改 docker的配置，重启搞定！
[image]

latel, 2024-04-08:
好滴，我试试先

latel, 2024-04-08:
好滴，我试试先

latel, 2024-04-08:
> 在你的磁盘那点-“修改”，然后拉到下面:如图，想增加多大就+几G，新建后格式化，然后挂载为 docker数据分区，对应修改 docker的配置，重启搞定！ ![5221712540881_ pic](https://private-user-images.githubusercontent.com/141570215/320333056-66abdbea-692c-4942-9208-5e884bf59c5d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI1NTUxOTksIm5iZiI6MTcxMjU1NDg5OSwicGF0aCI6Ii8xNDE1NzAyMTUvMzIwMzMzMDU2LTY2YWJkYmVhLTY5MmMtNDk0Mi05MjA4LTVlODg0YmY1OWM1ZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQwOFQwNTQxMzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMTA5M2M3MGJkZTRlZmIyNjk2YTgyM2FlY2E2NGI1YjNjYjY5NTU2MWIyZjljYzMwM2VjZjJkMjQxNDIzN2E1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-2R_0O-BVN1lUGYtbmmccOLBGpGFdIFrAxoME_hz3rA)

我这里是1507328加1024*1024*1024*5，15601663不动吗？

latel, 2024-04-19:
HELP 😢

oppen123, 2024-05-07:
自己创建一个然后把格式转换好，到挂在点挂在到overlay并重启系统，在此之前你需要先备份。

bboy2000, 2024-10-27:
刷好卡，在电脑上用diskgenius把未分配的空间格式化成ext4,然后克隆分区到这个大容量分区，克隆后删除原来的小的分区。

latel, 2024-10-28:
> 刷好卡，在电脑上用diskgenius把未分配的空间格式化成ext4,然后克隆分区到这个大容量分区，克隆后删除原来的小的分区。

多谢知道，这样说我就会了