stupidloud/nanopi-openwrt #1364: 终端内更新报错

latel, 2024-04-19:
```bash
# wget -qO- https://github.com/stupidloud/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | bash
Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Upgrading coreutils-truncate on root from 9.3-1 to 9.3-r1...
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/coreutils-truncate_9.3-r1_aarch64_generic.ipk
Configuring coreutils-truncate.
准备更新到
--2024-04-19 09:10:31--  https://mirror.ghproxy.com/https://github.com/stupidloud/nanopi-openwrt/releases/download//r4s.img.gz
Resolving mirror.ghproxy.com... 129.151.136.35
Connecting to mirror.ghproxy.com|129.151.136.35|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2024-04-19 09:10:32 ERROR 404: NOT FOUND.

gzip: invalid magic
awk: md5sum.txt: No such file or directory
root@OpenWrt:~# wget -qO- https://github.com/stupidloud/nanopi-openwrt/raw/master/scripts/autoupdate-bash.sh | bash
```

oppen123, 2024-05-07:
看样子好像是网络的问题

latel, 2024-05-10:
> 看样子好像是网络的问题

明显不是啊

AnzhiZhang, 2024-06-12:
也有这个问题，明显不是网络问题

chk970222, 2024-06-16:
不是说了吗？404

AnzhiZhang, 2024-06-16:
> […](#)

404 是因为服务端没有了，不是因为网的问题获取不到了