stupidloud/nanopi-openwrt #1371: 磁盘莫名其妙成了只读，任何设置和修改都无法保存

latel, 2024-05-13:
[image]

jimmieyangcsjs, 2024-06-19:
OP经常出这个问题  一般是磁盘满了

latel, 2024-06-19:
> OP经常出这个问题 一般是磁盘满了

没满，感觉是tf卡坏了触发写保护了

jimmieyangcsjs, 2024-06-19:
e2fsck /dev/mmcblk1p2
mount -o remount rw 
试试

freedombig, 2024-07-22:
我在这个目录也是只读 tmp 目录可以写

latel, 2024-07-28:
> 我今天莫名其妙卡死了，各种配置也是保存不了，在 root 目录也是只读， tmp 目录可以写。另外一台机器 root 目录可以写

可能是tf卡坏了

suckmyS, 2024-10-23:
系统默认开机会执行自启脚本/etc/rc.local，脚本里默认有/usr/bin/mount -o remount,rw /命令，开启执行一次使根目录可读写。但最近几个版本升级后发现/usr/bin/下没有了mount的二进制文件，只有/bin/下有mount，所以尝试一下拔卡，用linux的电脑或虚拟机挂在sd卡的根目录，挂载路由的根目录，然后找到/etc/rc.local，编辑将里面的/usr/bin/mount修改成//bin/mount应该就能解决问题。

latel, 2024-10-24:
原来是这样，我以为每次都是我tf卡又烧了，我现在换了带emmc的nanopi r5了 🤓

suckmyS, 2024-10-24:
估计是嵌入式的问题，如果是x86由grub引导直接赋予rw权限就不会出现这个问题> 原来是这样，我以为每次都是我tf卡又烧了，我现在换了带emmc的nanopi r5了 🤓

suckmyS, 2024-10-24:
估计是嵌入式的问题，如果是x86由grub引导直接赋予rw权限就不会出现这个问题

bboy2000, 2024-10-27:
> 系统默认开机会执行自启脚本/etc/rc.local，脚本里默认有/usr/bin/mount -o remount,rw /命令，开机执行一次使根目录可读写。但最近几个版本升级后发现/usr/bin/下没有了mount的二进制文件，只有/bin/下有mount，所以尝试一下拔卡，用读卡器将sd卡插上linux电脑或虚拟机上，挂载路由的根目录，然后找到/etc/rc.local，编辑将里面的/usr/bin/mount修改成//bin/mount应该就能解决问题。

最近那个版本正常点？我r1plus以前用这个也是很稳定，最新换成新的版本各种只读，卡死......