stupidloud/nanopi-openwrt #1384: openwrt能够获取ipv6地址，但test-ipv6无法通过

splxyang, 2024-11-16:
在PVE虚拟机里搭了OpenWrt，内核版本 6.1.83。
[image]

深圳电信，光猫设置了桥接。
[image]

OpenWrt设置拨号上网，网络正常，然后设置WAN6，可以正常获取IPV6-PD。
[image]

但是打开test-ipv6，显示检测不到ipv6地址。
[image]

用自带的traceroute去测试，连续返回相同的ip，对照之后正是WAN6口的ipv6。
[image]

有没有大佬能够指导一下，确实不懂，搞了半天了，拜托拜托~

stupidloud, 2024-11-17:
你的内网设备有没有获取到v6
[image]

splxyang, 2024-11-19:
我解决了，是我光猫设置有问题，设置了两个上网连接，一个是PPPoE封装，另一个是IPoE封装，似乎两个链接相互之间冲突了，原理不清楚，反正把IPoE改回PPPoE就可以了。